TrevorDecker/CMU_Mechatronics_2015_TeamB #5: Design Gripper

TrevorDecker, 2015-02-20:
Design the Gripper

stephaniechen456, 2015-02-20:
Origin: Weight
Description: Not enough area to grip for weight from robot and gearbox. Dynamic load from pivoting also needs to be considered

stephaniechen456, 2015-02-20:
Origin: Coupler not properly attached to motor shaft or threaded rod
Description: Motor shaft and bolt slip in the coupler, leading to no actuation

TrevorDecker, 2015-03-16:
Prototype 2 has been built, can currently hold about 5lb's 
plan to optimize between now and march 22nd If it can not hold 20ish lb we will move to a different redesign based heavily on a clamp.

ihartwig, 2015-04-02:
I think we really need to use the (immense) weight of our own robot to hold ourself up. A good example of this comes from the 2010 FIRST Robotics Challenge where the 120-150 lb robots had to lift themselves off the ground. [Many of the successful teams](http://www.adambots.com/wp-content/gallery/first/2010/robot/F10-Robot-01.jpg) just touched the pole on opposite sides at 2 points and used gravity to keep them at a fixed point on the pole. We can do the same by using oversized c-channel as our grippers, applying the rubber c channel on the edges as grip, and using the pivot gearbox to rotate the c channel into a binding position.

TrevorDecker, 2015-05-12:
Decided to use a piece of C channel that we found

closed 2015-05-12 by TrevorDecker