TrevorDecker/CMU_Mechatronics_2015_TeamB #6: Test Extending Arm

stephaniechen456, 2015-02-20:


stephaniechen456, 2015-02-20:
Origin: Width of arm and cleaning unit
Description: The width of the extending arm provides a moment that causes the robot to want to tip away from the window

stephaniechen456, 2015-02-20:
Origin: Joints
Description: There might be play in the bearing and bracket attachment on the two beams. This will cause torque at the joints and may affect our performance in moving linearly.

stephaniechen456, 2015-02-20:
Origin: Weight
Description: Mechanisms (motors, cleaning unit, drive shafts) weigh down extending arm, creating bending in undesirable places.

TrevorDecker, 2015-03-16:
Cad model finished need to machine parts and assemble system

TrevorDecker, 2015-04-03:
Fully assembled, and operational.  Still need to add some convince holes to allow for smother operation, and easier maintenance.

closed 2015-04-03 by TrevorDecker